waveform80/picamera #627: HQ camera error maximum resolution

paulinescho, 2020-05-20:
Hello,

I'm having troubles with the HQ camera and the capture with picamera library on python.
With command line I can capture an image with the new maximum resolution of the HQ camera (4056,3040) but if I try to ask that resolution on python with camera.resolution, I can't capture the picture, I have a message error "failed to enable component : ENOSPC" or "ENOMEM" but it work with lower resolution and with preview.
I've check the gpu memory and it's at 128.
I've also noticed that once I've had an error message on python, then I have a similar error message on command line. If I reboot the Rpi, the command line is working again.
I've run update and upgrade but I still have these problems.
I didn't had any issues with the camera module v2 before.

Does anyone have any idea of what's going on and how to correct it ?

Thank's

6by9, 2020-05-20:
Initially try increasing gpu_mem further - higher resolution images require more memory. Each input 12MPix image is also now 12bits/pixel instead of 10, so that makes them 18MB each. Output images at 12MPix are also 18MB (for YUV420, 24MB for YUV422, and 36MB for RGB).
128MB is only enough for 7 input or YUV420 images at that full res.

paulinescho, 2020-05-20:
Ok thank's. I've try 144 before witch changed nothink but I'll try higher.

paulinescho, 2020-05-20:
But if it was the problem, why can I take a 12MPix picture with command line ?

6by9, 2020-05-20:
Raspistill configures the camera for typically max 1080p preview, and a single stills capture buffer.
On the input side, 1x12MPix max res input buffer or 3x preview buffer sized frames, is generally 1x12MPix buffer. (18MB). 12bpp.
Output side likewise is 3x1080p YUV420 (3MB each), and one 12MPix YUV420 (18MB).
Total around 45MB.

Picamera asks for 12Mpix preview, so we now have 3 12MPix 12bpp images.
Likewise on the output side it wants 12MPix, so a further 3 12MPix YUV420 buffers, plus a stills buffer. And because of the way it moves buffers to/from the GPU, there will be additional buffers there. So a minimum of 7 internal 12MPix 12bit Bayer or YUV420 buffers, plus the ones to shift the images from the GPU. You've already exceeded 128MB.

paulinescho, 2020-05-20:
It work with gpu_mem at 176 !
Thank's a lot !

closed 2020-05-20 by paulinescho