waveform80/picamera #629: raw bayer support for HQ camera

scriptorron, 2020-06-15:
This is related to the HQ camera:
The `capture` function with `bayer=True` does not return the whole Bayer data and a second call to `capture` leads to a `PiCameraRuntimeError`.

The issue can be replicated with:
```python
from __future__ import (
    unicode_literals,
    absolute_import,
    print_function,
    division,
    )

import time
import io
import picamera

with picamera.PiCamera() as camera:
    # Let the camera warm up for a couple of seconds
    time.sleep(2)
    # Capture the image, including the Bayer data
    stream = io.BytesIO()
    camera.capture(stream, format='jpeg', bayer=True)
    print("Camera version: %s" % camera.exif_tags['IFD0.Model'])
    print("1st stream size: %d" % stream.tell())
    # try next capture
    time.sleep(2)
    stream = io.BytesIO()
    camera.capture(stream, format='jpeg', bayer=True)
    print("2nd stream size: %d" % stream.tell())
```

I set `DEBUG = 10` in `PiEncoder`to get more debug information.  The console outputs are:
```
pi@raspicam03:~/HubblePi $ python test.py
 vc.ril.camera [2]                             [0] vc.ril.image_encode [0] 
   encoding    OPQV-strips             OPQV-strips      encoding       JPEG
      buf      10x128          -->          10x128         buf         1x345600
    bitrate    0bps                           0bps       bitrate       0bps
     frame     720x480@30fps         720x480@30fps        frame        0x0@30fps
<MMALBuffer object: flags=______ command=none length=345600>
<MMALBuffer object: flags=______ command=none length=345600>
<MMALBuffer object: flags=______ command=none length=345600>
<MMALBuffer object: flags=_E___X command=none length=96465>
Camera version: RP_testc
1st stream size: 1133265
 vc.ril.camera [2]                             [0] vc.ril.image_encode [0]
   encoding    OPQV-strips             OPQV-strips      encoding       JPEG
      buf      10x128          -->          10x128         buf         1x345600
    bitrate    0bps                           0bps       bitrate       0bps
     frame     720x480@30fps         720x480@30fps        frame        0x0@30fps
Traceback (most recent call last):
  File "test.py", line 29, in <module>
    print("2nd stream size: %d" % stream.tell())
  File "/home/pi/HubblePi/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/home/pi/HubblePi/picamera/camera.py", line 752, in close
    raise exc
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```

The size of the first received stream fits to the sum of the buffer lengths. But this can not be the whole bayer pattern. I would expect about 20 MiByte.

A second call to `capture` ends in an error. This is certainly a subsequent fault.

I am running this on a RaspberryPi Zero WH with the latest Raspian:
```
pi@raspicam03:~/HubblePi $ uname -a
Linux raspicam03 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
```

When trying `raspistill --raw -o raspistill_HQ.jpg` I get a JPG file with 22 MiByte. I have not checked if the Bayer pattern is complete but just the size makes more sense to me. Furthermore on [https://www.raspberrypi.org/forums/viewtopic.php?t=273500](https://www.raspberrypi.org/forums/viewtopic.php?t=273500) the people use `raspistill` to produce raw Bayer data.

6by9, 2020-06-15:
Update your firmware.

There was an early bug in the firmware that allowed the use of the 10bit 720p mode for stills captures. That is a 1125kB Bayer image.
In doing so there was also an issue where it failed to switch back to the 12bit viewfinder mode correctly, which is why the pipeline times out for the second capture.

I've just run your script with latest firmware and got 
1st stream size: 19710134
2nd stream size: 19709944
both of which are reasonable for the 12MPix capture (18MB of raw data as 12bpp).

scriptorron, 2020-06-15:
Thank you very much for your quick answer!

I did `sudo apt-get update` and `sudo apt-get upgrade` but the firmware is still the same. How do I get an updated firmware? With `rpi-update`? Which firmware do you have running?

6by9, 2020-06-15:
> I did sudo apt-get update and sudo apt-get upgrade but the firmware is still the same. How do I get an updated firmware? With rpi-update? Which firmware do you have running?

I was expecting you to get the firmware bump from 1st June, but someone else had the same issue of apt not updating on https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=276556

```apt-cache policy raspberrypi-bootloader``` should list Candidate: 1.20200601-1
```vcgencmd version``` should return something from the same date.
I'm investigating whether something went wrong in the building of that Raspbian package.

I'm using debug firmware from the head of our internal tree, but this issue is fixed in rpi-update back in May.

6by9, 2020-06-15:
OK, I've had an answer - the Pi4 firmware got bumped, but not the Pi0-3 one.
If you're on Pi0-3, you'll have to use rpi-update for now too get the fix.

scriptorron, 2020-06-15:
After `rpi-update` everything works fine now. Many thanks for your great support! Have a nice day. Bye.

closed 2020-06-15 by scriptorron