waveform80/picamera #630: Can't install it on virtualenv on a raspberry pi 4 - (8GB)

digas, 2020-06-17:
Using python 3.7.3 
tried:
(cv) pi@opencv:~ $ pip install "picamera[array]"
Collecting picamera[array]
  Using cached picamera-1.13.tar.gz (143 kB)
Requirement already satisfied: numpy in ./.virtualenvs/cv/lib/python3.7/site-packages (from picamera[array]) (1.18.5)
Building wheels for collected packages: picamera
  Building wheel for picamera (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.virtualenvs/cv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-5puzdgaw
       cwd: /tmp/pip-install-1w_1xz0u/picamera/
  Complete output (54 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib
  creating build/lib/picamera
  copying picamera/frames.py -> build/lib/picamera
  copying picamera/renderers.py -> build/lib/picamera
  copying picamera/streams.py -> build/lib/picamera
  copying picamera/mmalobj.py -> build/lib/picamera
  copying picamera/array.py -> build/lib/picamera
  copying picamera/encoders.py -> build/lib/picamera
  copying picamera/exc.py -> build/lib/picamera
  copying picamera/bcm_host.py -> build/lib/picamera
  copying picamera/color.py -> build/lib/picamera
  copying picamera/display.py -> build/lib/picamera
  copying picamera/__init__.py -> build/lib/picamera
  copying picamera/camera.py -> build/lib/picamera
  copying picamera/mmal.py -> build/lib/picamera
  running egg_info
  writing picamera.egg-info/PKG-INFO
  writing dependency_links to picamera.egg-info/dependency_links.txt
  writing requirements to picamera.egg-info/requires.txt
  writing top-level names to picamera.egg-info/top_level.txt
  reading manifest file 'picamera.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  no previously-included directories found matching 'debian'
  no previously-included directories found matching 'docs'
  writing manifest file 'picamera.egg-info/SOURCES.txt'
  installing to build/bdist.linux-armv7l/wheel
  running install
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 145, in <module>
      main()
    File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 140, in main
      cmdclass             = {'install': CustomInstallCommand},
    File "/home/pi/.virtualenvs/cv/lib/python3.7/site-packages/setuptools/__init__.py", line 161, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/pi/.virtualenvs/cv/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 259, in run
      self.run_command('install')
    File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 109, in run
      raise ValueError('This system does not appear to be a Raspberry Pi')
  ValueError: This system does not appear to be a Raspberry Pi
  ----------------------------------------
  ERROR: Failed building wheel for picamera
  Running setup.py clean for picamera
Failed to build picamera
Installing collected packages: picamera
    Running setup.py install for picamera ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.virtualenvs/cv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-z_le02d2/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.virtualenvs/cv/include/site/python3.7/picamera
         cwd: /tmp/pip-install-1w_1xz0u/picamera/
    Complete output (18 lines):
    running install
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 145, in <module>
        main()
      File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 140, in main
        cmdclass             = {'install': CustomInstallCommand},
      File "/home/pi/.virtualenvs/cv/lib/python3.7/site-packages/setuptools/__init__.py", line 161, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-1w_1xz0u/picamera/setup.py", line 109, in run
        raise ValueError('This system does not appear to be a Raspberry Pi')
    ValueError: This system does not appear to be a Raspberry Pi
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.virtualenvs/cv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1w_1xz0u/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-z_le02d2/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.virtualenvs/cv/include/site/python3.7/picamera Check the logs for full command output.
(cv) pi@opencv:~ $

Any ideia ?

digas, 2020-06-17:
But version :
(cv) pi@opencv:~ $ pip install -Iv  "picamera[array]"==1.12

Just installed perfectly ...

Lefuneste83, 2020-06-24:
Facing the same issue installing The Spaghetti Detective plugin in Octoprint on Pi 4 with HQ camera (fresh install).

[https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/66](url)

I managed to solve it by first forcing install of the picamera library before installing the plugin.

**sudo -s
export READTHEDOCS=True
/home/pi/oprint/bin/python2 -m pip install picamera**

As this was suggested here :
[https://github.com/waveform80/picamera/issues/578#issuecomment-523997913](url)

Still there seems to be an issue with the picamera library with HQ camera on Pi 4 as my install is fresh out of the box Octoprint.

mxmaxime, 2020-07-21:
Hi,

I've the same issue, so I'm using the version 1.12 but I didn't have this issue before... A mate did an upgrade of my RPI and then... broke.