waveform80/picamera #631: Vulkan support?

AkdM, 2020-06-21:
Hey there.

I'm total noob on that subject, but I came across [this tweet](https://twitter.com/0martint/status/1274012749174013954) today and I was wondering if picamera could be more efficient/efficient processing?

Thanks 😁

Btw, the repo can be found here: https://github.com/Yours3lf/rpi-vk-driver

6by9, 2020-06-22:
It's going to make no difference at all. Almost all the camera image processing is done in dedicated hardware known as the ISP (Image Sensor Pipeline), not on the 3D hardware.

There is the potential to render preview images via GL/GLES/Vulkan, but that's outside the scope of Picamera.

AkdM, 2020-06-22:
Thanks for the clear answer @6by9!

closed 2020-06-22 by AkdM