waveform80/picamera #632: trouble capturing MAX_RESOLUTION images to memory

copterspace, 2020-07-13:
Simple code bellow I tried to capture picamera image with MAX_RESOLUTION first to file and then to memory:
```
import time
from picamera.array import PiRGBArray
from picamera import PiCamera
camera = PiCamera()
camera.resolution = camera.MAX_RESOLUTION
rawCapture = PiRGBArray(camera)
time.sleep(2)
camera.capture('img/testimage.jpg')
camera.capture(rawCapture,'bgr')
````

testimage saves ok with resolution 3280x2464
but on `camera.capture(rawCapture,'bgr')` I get:

```
/home/pi/.local/lib/python2.7/site-packages/picamera/encoders.py:544: PiCameraResolutionRounded: frame size rounded up from 3280x2464 to 3296x2464
  width, height, fwidth, fheight)))
mmal: mmal_vc_port_enable: failed to enable port vc.ril.camera:out:2(BGR3): ENOMEM
mmal: mmal_port_enable: failed to enable port vc.ril.camera:out:2(BGR3)(0x1e10060) (ENOMEM)
Traceback (most recent call last):
  File "test3.py", line 9, in <module>
    camera.capture(rawCapture,'bgr')
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/camera.py", line 1425, in capture
    encoder.close()
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/encoders.py", line 431, in close
    self.stop()
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/encoders.py", line 419, in stop
    self._close_output()
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/encoders.py", line 349, in _close_output
    mo.close_stream(output, opened)
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/mmalobj.py", line 371, in close_stream
    stream.flush()
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/array.py", line 238, in flush
    self.array = bytes_to_rgb(self.getvalue(), self.size or self.camera.resolution)
  File "/home/pi/.local/lib/python2.7/site-packages/picamera/array.py", line 127, in bytes_to_rgb
    'Incorrect buffer length for resolution %dx%d' % (width, height))
picamera.exc.PiCameraValueError: Incorrect buffer length for resolution 3280x2464
```
If I change `camera.resolution = (1280,1024)` - everything is OK.

How can I capture images with MAX_RESOLUTION  onto memory, not to file?

```

braingram, 2020-09-29:
The mmal: mmal_vc_port_enable: failed to enable port vc.ril.camera:out:2(BGR3): ENOMEM line makes me think you need to increase the gpu memory for the pi.

Your code works on a pi 4 with a high quality camera and "gpu_mem=512" in /boot/config.txt

See this page for information on setting the gpu memory size (note that the gpu_mem_512 option did not work for me and at least on a pi 4 with 2 GB setting gpu_mem=1024 left the pi unbootable).

6by9, 2020-09-30:
> See this page for information on setting the gpu memory size (note that the gpu_mem_512 option did not work for me and at least on a pi 4 with 2 GB setting gpu_mem=1024 left the pi unbootable).

Already stated in [the docs](https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md) that setting gpu_mem stupidly high will result in an unbootable Pi. And it's is being [updated again](https://github.com/raspberrypi/documentation/pull/1689) to be more explicit.

> Incorrect buffer length for resolution 3280x2464

I suspect that something is configuring the size incorrectly as the buffer coming back from the GPU is always padded to a multiple of 32 in width and 16 in height. 3280 is not a multiple of 32.
(bgr actually get more complex as it is 3bytes/pixel, so you get further odd multiples kicking around).