waveform80/picamera #640: Video stream does not maintain framerate when image is dark

courtarro, 2020-08-06:
I've found that the video stream does not maintain the desired framerate whenever the image is dark. This is in h.264 mode. I believe this is due to an internal buffer somewhere that only writes data to the output file/socket whenever a certain data size is reached.

Here's a simple test script:

```python
ALL_IFRAMES = False
OVERLAY_TIMESTAMP = False

import datetime
import socket
import picamera

try:
    camera = picamera.PiCamera()

    server_socket = socket.socket()
    server_socket.bind(('0.0.0.0', 8000))
    server_socket.listen(0)

    connection = server_socket.accept()[0].makefile('wb')

    camera.resolution = (1920, 1080)
    camera.framerate = 30
    camera.awb_mode = 'sunlight'
    camera.shutter_speed = 30000
    camera.iso = 100

    camera.start_recording(connection, format='h264', level='4.2', intra_period=1 if ALL_IFRAMES else None)

    if OVERLAY_TIMESTAMP:
        while True:
            camera.annotate_text = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S.%f')
            camera.wait_recording(0.01)
    else:
        camera.wait_recording(5000000)
        
except (KeyboardInterrupt, ConnectionResetError, BrokenPipeError):
    pass
finally:
    if camera:
        try:
            print('Stopping recording')
            camera.stop_recording()
        except:
            pass
    
        print('Closing camera')
        camera.close()

    if connection:
        try:
            print('Closing client connection')
            connection.close()
        except:
            pass

    if server_socket:
        print('Closing server socket')
        server_socket.close()
```

On another machine, run `mplayer` with a very short minimum buffer size so that it starts quickly. In this example, the server above is running on `192.168.4.1`.

```bash
nc 192.168.4.1 8000 | mplayer -fps 120 -cache 1024 -cache-min 1 -
```

Note that I tell `mplayer` to try an fps of 120 so that it will play the video as fast as possible (to keep latency down).

# Symptom

When very little is changing in the output video stream, data is not being sent to the receiver. If you enable the timestamp, the problem improves slightly, but forcing all I-frames mostly works around the issue. Similarly, higher ISO values like 800 tend not to have the problem. Again, I theorize this is due to the noise adding to frame complexity and thus reaching the buffer flush point sooner.

Is it possible to determine where the output buffer latency is coming from?

JoshuaRack, 2020-08-07:
Have you tried setting exposure_mode = ''fixedfps' ?

6by9, 2020-08-08:
Sounds like you're hitting a feature of TCP/IP - https://en.wikipedia.org/wiki/Nagle%27s_algorithm

courtarro, 2020-08-08:
> Sounds like you're hitting a feature of TCP/IP - https://en.wikipedia.org/wiki/Nagle%27s_algorithm
> What size are each of the encoded frames when they hit the other end?

I think you're right. I saved some of the stream and the black frames are tiny: some as small as 200 bytes, with many around 1000 bytes. The keyframes are about 60kb. Doing more testing, the main problem is aggregation of data within the sending socket before pushing it over the wire. I tried `TCP_NODELAY`, as suggested, but that only partially solves the problem. I actually got the best result by doing the following:

```python
class FlushingBufferProxy:
    def __init__(self, outgoing_socket):
        self._outgoing_socket = outgoing_socket
    
    def write(self, b):
        self._outgoing_socket.write(b)
        self._outgoing_socket.flush()
```

Then, I just hand the proxy to picamera:

```python
camera.start_recording(FlushingBufferProxy(connection))
```

This greatly reduces the latency, though I'm not 100% sure where the `flush()` call is having its main effect: within Python? the Linux kernel? the network card? All of the above?

There is a secondary issue which is that mplayer has a few thresholds before it will show video. One is `-cache-min`, which sets the percent size of the buffer that must be filled before it will attempt to decode/display anything. I use `-cache-min 1` (which means the 1MB cache must be 1% full before any attempt to decode, or about 10kb of data). However, there seems to be another threshold of minimum received payload size before it will estimate (and determine) that the incoming data contains an h.264 elementary stream. I would like to find a way to tell mplayer that the video is H264-ES manually, to bypass that delay, but I haven't yet found a way to do that.

In any case, none of this is really picamera's fault, so y'all are welcome to close this issue report unless the ongoing conversation is useful.

courtarro, 2024-01-09:
Situation documented ... not really a picamera issue. Closing.

closed 2024-01-09 by courtarro