waveform80/picamera #642: Exposure Mode Setting

Charliebegood, 2020-09-07:
Hi!

I can't get the exposure mode to work. It seems that the camera isn't able to settle and properly adjust its settings before capturing the picture. 

Here's the code:
[image]

Output straight from the code:
[image]

Output using the following raspistill command:
`raspistill -w 640 -h 360 --exposure "night" -o "/home/pi/Desktop/raspistill.jpg"`

[image]

Any help is welcome! Thanks 🙌

6by9, 2020-09-07:
Note all the examples in the [docs](https://picamera.readthedocs.io/en/latest/recipes1.html) include a call to ```start_preview``` before waiting for things to warm up.
AFAIK setting the properties as you've done will not start the camera, therefore the sleep is doing nothing except wasting time.

Charliebegood, 2020-09-07:
I had already tried with the `start_preview` but it produces the same output.

Code tested:
[image]

Output:
[image]