waveform80/picamera #646: Exposure time longer than 1/2 second when Shutter_speed = 0

jaazza, 2020-09-23:
Hello, 

I'm struggling to obtain a shutter speed greater than 1/2 second, when other variables are set. Currently, my settings are as follows: 

`import picamera
p = picamera.PiCamera()
p.resolution = (4056, 3040)
p.sensor_mode =3
p.shutter_speed = 0
p.iso = 0
p.exposure_mode = 'night'
sleep =(10)`

I'm running the camera (HQ) remotely wanting to capture at all times of day and night. If I'm reading the documentation correctly, mode 3 should allow for 1 second+ exposures, and 'night' exposure mode should as well. When using the above settings the result is an iso of 320 and exposure time of 1/2 second. I can force a longer exposure by utilizing shutter_speed = 1000000 or similar, but then there are overexposure issues during the day. 

Any guidance would be appreciated. Thank you.

6by9, 2020-09-23:
The maximum exposure time in the ```night``` profile is 500ms - that's how the tuning file is configured.
Longer exposure times are only available with manually set times via shutter_speed.

Using very long exposures in automatic modes confuses people as their preview updates slower and slower, hence the limit.

jaazza, 2020-09-23:
> The maximum exposure time in the `night` profile is 500ms - that's how the tuning file is configured.
> Longer exposure times are only available with manually set times via shutter_speed.
>
> Using very [...]

Thank you. Is there a document that outlines max exposure times for the other modes? I cant seem to find the data.

joaquinabian, 2021-02-11:
I cannot get more than 99.99 ms exposure with the HQ camera with **_sensor mode = 3_** and **_exposure_mode = 'night'_**
How can I arrive to the 500 ms you comment here ? 

For example, with the lens covered with its cap, this code:

```
#!/usr/bin/python3
#
from picamera import PiCamera
import os
import glob
from time import sleep
from fractions import Fraction
#
path ='/home/pi/timelapse_test'
photo = os.path.join(path, 'image{0:06d}.jpg')
report = '%i XS=%7i FR=%7.3f'
#
if os.path.exists(path):
    path_glob = os.path.join(path, '*.jpg')
    for arch in glob.glob(path_glob):
        os.remove(arch)
else:
    os.mkdir(path)
#    
cam = PiCamera()
#
cam.sensor_mode = 3 
cam.iso = 1600
cam.framerate = Fraction(5, 1000)
cam.exposure_mode = 'night'
sleep(20)
#
print('XM= ', cam.exposure_mode)
print('SM= ', cam.sensor_mode)
print('CI= ', cam.iso)
print('CR= ', cam.resolution)
#
for i in range(10):
    print(report % (i, cam.exposure_speed, cam.framerate))
    cam.capture(photo.format(i))
    sleep(10)
```
produces:

```
XM=  night
SM=  3
CI=  1600
CR=  1920x1080
0 XS=  99990 FR=  0.005
1 XS=  99990 FR=  0.005
2 XS=  99990 FR=  0.005
.........................
```