waveform80/picamera #652: Install picamera on a non-RPi machine

Sanazjd, 2020-10-30:
I want to run picamera on a server (Ubuntu machine) that orders clients (8 raspberry pis) to capture images simultaneously. For this, I need to install picamera on the server which is not a raspberry pi and do not have the Pi camera. Conventional ways of installation like "pip3 install picamera" is not working obviously, with this error

ValueError: Unable to determine if this system is a Raspberry Pi

 Is there any way to install picamera on a non-RPi machine?

Esser420, 2020-11-06:
You cannot install picamera on other machines and it doesn't seem necessary for your use case. Installing it on the server means you would have a camera connected to your server which is not what you want I think.
You have just need to have it installed on the raspberry pis and your server sends commands the network.

optimuspaul, 2020-12-16:
I have a use case like this where I'm trying to install `picamera` in a docker image that does get deployed on RPi's. Unfortunately I have to build on an actual RPi whereas all of the other parts of my codebase can be built on other hardware.

Sanazjd, 2020-12-16:
I installed it on all the RPis and just send the command through server. I was worried about the synchronization, because I need to run other stuff tied to imaging, but it worked.