waveform80/picamera #656: Exposure compensation at 25 or -25 is same as 0

dustinkerstein, 2020-11-20:
On my Pi HQ camera, setting exposure compensation works for -25 < values < 25, but when set to 25 or -25 it behaves the same as 0. Are other folks able to replicate that? Maybe it's just a documentation issue, but I figured I'd reach out.

Also, a question: It seems that setting exposure compensation uses both analog+digital gain, rather changing shutter speeds within the available ranges defined by current FPS. Is there a way to force the camera to use shutter speed compensation rather than gain? Thanks!

6by9, 2020-11-20:
EV range is +/-4 stops in 1/6s, therefore -24 to +24. 

EV adjusts the target illumination for the scene. Generally AE/AGC will increase shutter speed first within the defined range (generally the max is determined by the frame rate), and then increases gain.
If shutter speed is at the max then it can't be increased further.

dustinkerstein, 2020-11-20:
Got it. That makes sense for -24 to +24. It's a little confusing in the [docs](https://picamera.readthedocs.io/en/release-1.10/api_camera.html#picamera.camera.PiCamera.exposure_compensation) which say it can be -25 or +25 when queried, but doesn't explicitly state that when setting it needs to be -24 to +24. Should that be updated?

Thanks for describing the logic. Makes sense.