waveform80/picamera #661: What is the correct bayer format?

xlla, 2020-12-08:
In document [here](https://picamera.readthedocs.io/en/release-1.13/recipes2.html#bayer-data), 
the data layout seems
<pre>
byte 1   byte 2   byte 3   byte 4   byte 5
 AAAAAAAA BBBBBBBB CCCCCCCC DDDDDDDD AABBCCDD
</pre>

In this example [bayer_data.py](https://github.com/waveform80/picamera/blob/master/docs/examples/bayer_data.py), line 61, it said data layout is 
<pre>
 byte 1   byte 2   byte 3   byte 4   byte 5
 AAAAAAAA BBBBBBBB CCCCCCCC DDDDDDDD DDCCBBAA
</pre>

6by9, 2020-12-08:
Same as the V4L2 formats
https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/pixfmt-srggb10p.html
which I think implies the second option.

Also note #462

xlla, 2020-12-09:
@6by9  Thanks a lot! I got it! But I am still confused with other aspect.
1  For imx477, does it obey the same rules?
 <pre>
 byte 1   byte 2   byte 3   
 AAAAAAAA BBBBBBBB  BBBBAAAA
</pre>

2 Does HQ camera have the same bayer layout or with vflip/hflip/etc ?
<pre>
#
# GBGBGBGBGBGBGB
# RGRGRGRGRGRGRG
# GBGBGBGBGBGBGB
# RGRGRGRGRGRGRG
#
</pre>

I have run example bayer_data.py successful on pi4 with hq camera, but the image looks too purple while open the image.data

3 Ref to this [documents](https://www.sony-semicon.co.jp/products/common/pdf/IMX477-AACK_Flyer.pdf),  only "Full(4:3) Normal 40 fps" use 12 bit ADC, but the amended example [bayer_data.py](https://github.com/xlla/picamera/blob/master/docs/examples/bayer_data.py#L55) decode data using 12bit algorithm successful in default settings, I think it's default is not 40fps .

6by9, 2020-12-09:
HQ camera is 12bpp, so follows the 12bit packing 
https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/pixfmt-srggb12p.html
V4L2 has descriptions for all the CSI2 packed formats.

The Bayer order does change based on sensor and flips on. The best approach is to read it (and the bit depth) out of the header, same as the dcraw fork does - https://github.com/6by9/dcraw/blob/master/dcraw.c#L6463

xlla, 2020-12-10:
@6by9  Thank you very much, I have learned a lot from you!

closed 2020-12-10 by xlla