waveform80/picamera #672: Failed to create MMAL component b'vc.camera_info': Function not implemented

fitzgeraldsystems, 2021-02-01:
- After I installed libraspberrypi-bin (and change permissions to /dev/vchiq/) I was able to use `$ raspistill -o output.jpg` without error. I did not build userland from source so I think this error is different to [Error #334](https://github.com/waveform80/picamera/issues/334) and [Error #540](https://github.com/waveform80/picamera/issues/540).
- I used pip (in virtual environment) to install picamera and have version 1.13.
- I attempt to run a simple script that calls the PiCamera class and I get the following error

```
$ python3 cameratest.py
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):`
  File "cameratest.py", line 4, in <module>
    camera = PiCamera()
  File "/home/ubuntu/environments/test_env/lib/python3.8/site-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/home/ubuntu/environments/test_env/lib/python3.8/site-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/home/ubuntu/environments/test_env/lib/python3.8/site-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/home/ubuntu/environments/test_env/lib/python3.8/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
```
```
$ uname -a
Linux ubuntu 5.4.0-1028-raspi #31-Ubuntu SMP PREEMPT Wed Jan 20 11:30:45 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```
I'm running ubuntu 20.04 server on RPi4
```
$ vcgencmd get_camera
supported=1 detected=1
```
```
$ ldconfig -p | grep libmmal
	libmmal_vc_client.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libmmal_vc_client.so
	libmmal_util.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libmmal_util.so
	libmmal_core.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libmmal_core.so
	libmmal_components.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libmmal_components.so
	libmmal.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libmmal.so
```
```
$ ldd /lib/aarch64-linux-gnu/libmmal.so
	linux-vdso.so.1 (0x0000ffffa48de000)
	libmmal_core.so => /lib/aarch64-linux-gnu/libmmal_core.so (0x0000ffffa4879000)
	libmmal_util.so => /lib/aarch64-linux-gnu/libmmal_util.so (0x0000ffffa4858000)
	libvcos.so => /lib/aarch64-linux-gnu/libvcos.so (0x0000ffffa483c000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffffa46c9000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffffa48ae000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffffa4699000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffffa4685000)
```
Please advise on how to solve the `mmal: mmal_component_create_core: could not find component 'vc.camera_info'` error

6by9, 2021-02-01:
MMAL isn't currently supported on a 64bit userspace.

Ubuntu have taken some patches for 64bit userspace that were reverted due to observing issues with them. That decision really falls on them to support.

fitzgeraldsystems, 2021-02-01:
@6by9 thanks. I'm not familiar with this domain but if 64bit processor supports 64bit and 32bit then couldn't I (somehow) build MMAL for 32bit?

6by9, 2021-02-01:
Generally multiarch isn't supported. It's possible, but not simple.

fitzgeraldsystems, 2021-02-04:
Ok, thanks! I guess then I have to wait till they update the firmware...

slowrunner, 2021-06-20:
Same exact issue, same kernel, now four months passed - any hopes for soon?

Where should we subscribe to be alerted to the fix when it happens?

Reason this is a big issue:  The ROS community is pushing for adoption of ROS2 (Foxy LTS) which requires Ubuntu Server 20.04 LTS.  ROS 2 robots want to use Python3.8 with picamera to "see".

l33klin, 2021-11-16:
Is there any update on this issue?

slowrunner, 2021-11-16:
It seems we all need to learn to use libcamera and the standard V4L libraries.

https://www.raspberrypi.com/news/an-open-source-camera-stack-for-raspberry-pi-using-libcamera/