waveform80/picamera #678: picamera to match raspistill

johnedstone, 2021-02-20:
Thank you for developing this code - awesome!
Why do the default settings for PiCamera not match the default settings for the command `raspistill`?  I see that others have documented this as well.  What is happening under the covers and is there a flag that can be set so that they are the same?  **_The command raspistill always looks richer_**.  One can run `exif` and see that in general the ISO, the Shutter Speed, the Exposure Time, and the Brightness ususually differ.
Thanks for thinking about this.

Here is an example
`raspistill -w 1920 -h 1080 -o /var/www/html/pi_images/raspi_001.jpg`
compared to this block
```
import time
import logging
from picamera import PiCamera
with PiCamera() as camera:
    camera.resolution = (1920, 1080)
    #camera.resolution = (4056, 3040)
    #camera.resolution = (3280, 2464)

    camera.start_preview()
    time.sleep(2)

    try:
        for i, filename in enumerate(camera.capture_continuous(
                '/var/www/html/pi_images/picam_{timestamp:%H%M%S}-{counter:03d}.jpg')):
            logging.info(filename)
            if i == 1:
                break
            time.sleep(9)
    finally:
        camera.stop_preview()

```
And comparing the two
```
sdiff <(exif /var/www/html/pi_images/raspi_001.jpg) <(exif /var/www/html/pi_images/picam_*001.jpg )
EXIF tags in '/var/www/html/pi_images/raspi_001.jpg' ('Motoro |	EXIF tags in '/var/www/html/pi_images/picam_132148-001.jpg' (
--------------------+----------------------------------------	--------------------+----------------------------------------
Tag                 |Value					Tag                 |Value
--------------------+----------------------------------------	--------------------+----------------------------------------
Image Width         |1920					Image Width         |1920
Image Length        |1080					Image Length        |1080
Manufacturer        |RaspberryPi				Manufacturer        |RaspberryPi
Model               |RP_imx477					Model               |RP_imx477
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Date and Time       |2021:02:20 13:27:16		      |	Date and Time       |2021:02:20 13:21:48
YCbCr Positioning   |Centered					YCbCr Positioning   |Centered
Image Width         |64						Image Width         |64
Image Length        |48						Image Length        |48
Compression         |JPEG compression				Compression         |JPEG compression
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Exposure Time       |1/102 sec.				      |	Exposure Time       |1/106 sec.
Exposure Program    |Aperture priority				Exposure Program    |Aperture priority
ISO Speed Ratings   |40						ISO Speed Ratings   |40
Exif Version        |Exif Version 2.2				Exif Version        |Exif Version 2.2
Date and Time (Origi|2021:02:20 13:27:16		      |	Date and Time (Origi|2021:02:20 13:21:48
Date and Time (Digit|2021:02:20 13:27:16		      |	Date and Time (Digit|2021:02:20 13:21:48
Components Configura|Y Cb Cr -					Components Configura|Y Cb Cr -
Shutter Speed       |6.67 EV (1/101 sec.)		      |	Shutter Speed       |6.72 EV (1/105 sec.)
Brightness          |2.56 EV (20.20 cd/m^2)		      |	Brightness          |2.63 EV (21.21 cd/m^2)
Metering Mode       |Center-weighted average			Metering Mode       |Center-weighted average
Flash               |Flash did not fire				Flash               |Flash did not fire
Maker Note          |342 bytes undefined data			Maker Note          |342 bytes undefined data
FlashPixVersion     |FlashPix Version 1.0			FlashPixVersion     |FlashPix Version 1.0
Color Space         |sRGB					Color Space         |sRGB
Pixel X Dimension   |1920					Pixel X Dimension   |1920
Pixel Y Dimension   |1080					Pixel Y Dimension   |1080
Exposure Mode       |Auto exposure				Exposure Mode       |Auto exposure
White Balance       |Auto white balance				White Balance       |Auto white balance
Interoperability Ind|R98					Interoperability Ind|R98
--------------------+----------------------------------------	--------------------+----------------------------------------
EXIF data contains a thumbnail (24576 bytes).			EXIF data contains a thumbnail (24576 bytes).
```
and
```
>>> from pkg_resources import require
>>> print(require('picamera')[0].version)
1.13
```
and
```
lsb_release -a
Noo LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

johnedstone, 2021-02-20:
[image]
[image]
Uploaded two files to compare (see sdiff above as well)

johnedstone, 2021-02-20:
Noting that the command `raspistill` default has a 5 second delay, `Time delay 5000` then extending the `sleep` to 5 seconds after the `camera.start_preview()` does not help.  That is, the default `raspistill` command is still more rich than the default PiCamera class.

```
$ raspistill --verbose -w 1920 -h 1080 -o /var/www/html/pi_images/raspi_001.jpg

"raspistill" Camera App (commit 4a0a19b88b43 Tainted)

Camera Name imx477
Width 1920, Height 1080, filename /var/www/html/pi_images/raspi_001.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file /var/www/html/pi_images/raspi_001.jpg
Enabling encoder output port
Starting capture -1
Finished capture -1
Closing down
Close down completed, all components disconnected, disabled and destroyed

$ 
$ egrep -A 1 ' camera.start_preview'  continuous_picamera.py 
    camera.start_preview()
    time.sleep(5)
$ 
$ python --version
Python 3.7.3
$ 
$ python  continuous_picamera.py 
INFO:root:/var/www/html/pi_images/picam_143210-001.jpg
$ 
$ sdiff <(exif /var/www/html/pi_images/raspi_001.jpg) <(exif /var/www/html/pi_images/picam_*001.jpg )
EXIF tags in '/var/www/html/pi_images/raspi_001.jpg' ('Motoro |	EXIF tags in '/var/www/html/pi_images/picam_143210-001.jpg' (
--------------------+----------------------------------------	--------------------+----------------------------------------
Tag                 |Value					Tag                 |Value
--------------------+----------------------------------------	--------------------+----------------------------------------
Image Width         |1920					Image Width         |1920
Image Length        |1080					Image Length        |1080
Manufacturer        |RaspberryPi				Manufacturer        |RaspberryPi
Model               |RP_imx477					Model               |RP_imx477
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Date and Time       |2021:02:20 14:31:39		      |	Date and Time       |2021:02:20 14:32:10
YCbCr Positioning   |Centered					YCbCr Positioning   |Centered
Image Width         |64						Image Width         |64
Image Length        |48						Image Length        |48
Compression         |JPEG compression				Compression         |JPEG compression
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Exposure Time       |1/199 sec.				      |	Exposure Time       |1/168 sec.
Exposure Program    |Aperture priority				Exposure Program    |Aperture priority
ISO Speed Ratings   |40						ISO Speed Ratings   |40
Exif Version        |Exif Version 2.2				Exif Version        |Exif Version 2.2
Date and Time (Origi|2021:02:20 14:31:39		      |	Date and Time (Origi|2021:02:20 14:32:10
Date and Time (Digit|2021:02:20 14:31:39		      |	Date and Time (Digit|2021:02:20 14:32:10
Components Configura|Y Cb Cr -					Components Configura|Y Cb Cr -
Shutter Speed       |7.64 EV (1/199 sec.)		      |	Shutter Speed       |7.39 EV (1/167 sec.)
Brightness          |2.62 EV (21.06 cd/m^2)			Brightness          |2.62 EV (21.06 cd/m^2)
Metering Mode       |Center-weighted average			Metering Mode       |Center-weighted average
Flash               |Flash did not fire				Flash               |Flash did not fire
Maker Note          |342 bytes undefined data			Maker Note          |342 bytes undefined data
FlashPixVersion     |FlashPix Version 1.0			FlashPixVersion     |FlashPix Version 1.0
Color Space         |sRGB					Color Space         |sRGB
Pixel X Dimension   |1920					Pixel X Dimension   |1920
Pixel Y Dimension   |1080					Pixel Y Dimension   |1080
Exposure Mode       |Auto exposure				Exposure Mode       |Auto exposure
White Balance       |Auto white balance				White Balance       |Auto white balance
Interoperability Ind|R98					Interoperability Ind|R98
--------------------+----------------------------------------	--------------------+----------------------------------------
EXIF data contains a thumbnail (24576 bytes).			EXIF data contains a thumbnail (24576 bytes).
```

johnedstone, 2021-02-20:
Note: Using the following.
- [Raspberry Pi High Quality Camera](https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/)
- [Official Raspberry Pi HQ Camera Lens - Telephoto](https://www.canakit.com/raspberry-pi-hq-camera-16mm-telephoto-lens.html)

johnedstone, 2021-02-20:
Okay, I think I've found it.  Does this makes sense?
These two changes from the original PiCamera code posted above appear to match what raspistill is doing

- `sleep(5)` after `camera.start_preview()`
- and change `camera.start_preview()` to `camera.start_preview(resolution=(1024, 768))` as it appears that for PiCamera the preview resolution is fullscreen by default, [link](https://picamera.readthedocs.io/en/release-1.13/api_renderers.html#pipreviewrenderer), as opposed to raspistill, which from the `--verbose` output above (`Full resolution preview No`) it appears not to be fullscreen preview.

Visually the images appear to be close and `exif` appears very close.  I need to confirm in low light - stay tuned.
```$ sdiff <(exif /var/www/html/pi_images/raspi*_001.jpg) <(exif /var/www/html/pi_images/picam_*001.jpg )
EXIF tags in '/var/www/html/pi_images/raspi_001.jpg' ('Motoro |	EXIF tags in '/var/www/html/pi_images/picam_154552-001.jpg' (
--------------------+----------------------------------------	--------------------+----------------------------------------
Tag                 |Value					Tag                 |Value
--------------------+----------------------------------------	--------------------+----------------------------------------
Image Width         |1920					Image Width         |1920
Image Length        |1080					Image Length        |1080
Manufacturer        |RaspberryPi				Manufacturer        |RaspberryPi
Model               |RP_imx477					Model               |RP_imx477
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Date and Time       |2021:02:20 15:45:35		      |	Date and Time       |2021:02:20 15:45:52
YCbCr Positioning   |Centered					YCbCr Positioning   |Centered
Image Width         |64						Image Width         |64
Image Length        |48						Image Length        |48
Compression         |JPEG compression				Compression         |JPEG compression
X-Resolution        |72						X-Resolution        |72
Y-Resolution        |72						Y-Resolution        |72
Resolution Unit     |Inch					Resolution Unit     |Inch
Exposure Time       |1/132 sec.				      |	Exposure Time       |1/131 sec.
Exposure Program    |Aperture priority				Exposure Program    |Aperture priority
ISO Speed Ratings   |40						ISO Speed Ratings   |40
Exif Version        |Exif Version 2.2				Exif Version        |Exif Version 2.2
Date and Time (Origi|2021:02:20 15:45:35		      |	Date and Time (Origi|2021:02:20 15:45:52
Date and Time (Digit|2021:02:20 15:45:35		      |	Date and Time (Digit|2021:02:20 15:45:52
Components Configura|Y Cb Cr -					Components Configura|Y Cb Cr -
Shutter Speed       |7.04 EV (1/131 sec.)		      |	Shutter Speed       |7.03 EV (1/130 sec.)
Brightness          |2.61 EV (20.92 cd/m^2)		      |	Brightness          |2.60 EV (20.77 cd/m^2)
Metering Mode       |Center-weighted average			Metering Mode       |Center-weighted average
Flash               |Flash did not fire				Flash               |Flash did not fire
Maker Note          |342 bytes undefined data			Maker Note          |342 bytes undefined data
FlashPixVersion     |FlashPix Version 1.0			FlashPixVersion     |FlashPix Version 1.0
Color Space         |sRGB					Color Space         |sRGB
Pixel X Dimension   |1920					Pixel X Dimension   |1920
Pixel Y Dimension   |1080					Pixel Y Dimension   |1080
Exposure Mode       |Auto exposure				Exposure Mode       |Auto exposure
White Balance       |Auto white balance				White Balance       |Auto white balance
Interoperability Ind|R98					Interoperability Ind|R98
--------------------+----------------------------------------	--------------------+----------------------------------------
EXIF data contains a thumbnail (24576 bytes).			EXIF data contains a thumbnail (24576 bytes).
```

johnedstone, 2021-02-20:
It appears that the `raspistill` preview resolution is documented [here](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiPreview.c) as 1024x768 which supports the above observation that the setting for PiCamera `camera.start_preview(resolution=(1024, 768))` matches `raspistill`

And it appears that the first `sleep` need not be 5 seconds, but rather 1 - 2 seconds is enough.  That is to say that I believe setting `camera.start_preview(resolution=(1024, 768))` is the only adjustment to the Picamera class in order for it to output a nearly identical image as the default settings of `raspistill`

johnedstone, 2021-02-20:
Note: the above appears **_not_** to be true for low light.  Something else besides preview resolution is going on in low light.  So the question is still open: "What else besides setting the preview resolution to 1024x768 in PiCamera is needed to make it match `raspistill`?

johnedstone, 2021-02-21:
In order to get both bright and low light settings to approximate what raspistill is doing by default, then adding `exposure_mode = nightpreview` seems to accomplish this.  Listed below is the code that appears to work in both bright and dim light settings in order to approximate the default settings for raspistill

```
import time
import logging
import picamera

from datetime import datetime
from picamera import PiCamera

DEBUG = True

if DEBUG:
    logging.basicConfig(level=logging.INFO)
else:
    logging.basicConfig(level=logging.WARNING)

with PiCamera() as camera:
    '''
    These settings approximate the raspistill default in bright and dim light
        preview resolution: 1024x768
        exposure mode: nightpreview (adjusts in low light)
    '''

    camera.resolution = (1920, 1080)
    camera.exposure_mode = 'nightpreview'

    if camera.resolution == (1920, 1080):
        camera.annotate_text_size = 24 # default: 32, range: 6-160
    if camera.resolution == (4056, 3040):
       camera.annotate_text_size = 72

    # camera.annotate_background = picamera.Color('Black')
    camera.annotate_foreground = picamera.Color('White')

    camera.start_preview(window=(0,0,1024,768), resolution=(1024,768))
    time.sleep(2)

    try:
        for i, filename in enumerate(camera.capture_continuous(
                '/var/www/html/pi_images/picam_{timestamp:%Y%H%M%S}-{counter:03d}.jpg')):
            logging.info(filename)
            camera.annotate_text = 'frame #{}: {}'.format(i + 2, datetime.now())

            logging.info('interator: {}'.format(i))
            if i == 6:
                break
            time.sleep(9.5)
    finally:
        camera.stop_preview()
```

This is tested with this command, creating 10 sec interval images for 1 min.  The images are compared visually as well as with `exif`
```
/usr/bin/raspistill --verbose -w 1920 -h 1080 -t 60000 -tl 10000 -o /var/www/html/pi_images/frame%04d.jpg && python continuous_picamera.py
```

kif, 2022-06-09:
Nice piece of information: 
I was trying to make very long timelapses with exposure smoothing, switching from `exposure_mode = "auto"` to measure automatically the exposure (and calculating the Ev value knowing the sensor, the lens aperture and all gains and speed) to "off" with settings forced to reasonable values which are smoothed over a fairly long period of time (minutes).

Replacing "auto" with "nightpreview" gives a better/faster update on the setting when the camera is on standby-mode (i.e. between two shoots).
Thanks for the tip !

johnedstone, 2022-06-10:
Your welcome!