waveform80/picamera #679: How do I format the coordinates for the Exif?

BAChapin, 2021-02-22:
I just got my first Raspberry Pi product last week. I got a Raspberry Pi Zero W, and a PiCamera. I created a program (to be ran each day by crontab) that will take a photo, store the photo locally, transfer a copy of the photo to a local server, make a CSV log of the photo, and make an easily readable text daily report. I've gotten almost everything working correctly, but I have one particular issue I can't seem to figure out.

I'm using the Python PiCamera library to capture the photos everyday, and I wanted to set some static GPS data to each photo's EXIF metadata. (The GPS data needs to be static because the Pi is suction cupped to a window and wont be moving for at least a year) The issue I need help with is that I do not know how EXACTLY I need to format my GPS data for PiCamera's exif_tag property. Here are some of the formats I've tried and have not given me appropriate results:

**Attempt 1 - Float**
```
camera.exif_tags["GPS.GPSLatitude"] = 41.1027
camera.exif_tags["GPS.GPSLongitude"] = -85.1362
```

I get this error:

```
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    camera.capture(local_path)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1418, in capture
    encoder.start(output)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1125, in start
    self._add_exif_tag(tag, value)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 1097, in _add_exif_tag
    ct.sizeof(mmal.MMAL_PARAMETER_EXIF_T) + len(tag) + len(value) + 1)
TypeError: object of type 'float' has no len()
```

**Attempt 2 - String**

```
camera.exif_tags["GPS.GPSLatitude"] = "41.1027"
camera.exif_tags["GPS.GPSLongitude"] = "-85.1362"
```

No error is produced when running this code, however the output data from the exif after the image is captured is still inaccurate:

```
Latitude            |0.040
Longitude           |3162715.1775
```

**Follow Up Attempts - Other Strings**

```
# Style 1
camera.exif_tags["GPS.GPSLatitude"] = "(41) (6) (10.711)"
camera.exif_tags["GPS.GPSLongitude"] = "(85) (8) (8.968)"

# This produces
Latitude            |1.00
Longitude           |1.00

# Style 2
camera.exif_tags["GPS.GPSLatitude"] = "41 6 10.711"
camera.exif_tags["GPS.GPSLongitude"] = "85 8 8.968"

# This produces
Latitude            |6.8
Longitude           |10.6

# Style 3
camera.exif_tags["GPS.GPSLatitude"] = "41, 6, 10.711"
camera.exif_tags["GPS.GPSLongitude"] = "85, 8, 8.968"

#This produces
Latitude            |6.8, 0.014, 1.000
Longitude           |10.6, 0.008, 2.141837875
```

So, in short, can anyone help me by providing me with the means to format my strings? I am at a complete loss. The documentation for PiCamera only lists the different tags but does not explain how to format each of them.

6by9, 2021-02-22:
Compare against the Raspistill code for setting GPS coords?
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L1299

BAChapin, 2021-02-23:
> Compare against the Raspistill code for setting GPS coords?
> https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L1299

I've never learned C and my C++ skills are a bit rusty. But looking it over gave me an idea, and I worked it out. Thank you for your help!

closed 2021-02-23 by BAChapin

perezzini, 2021-09-03:
Hi, @BAChapin , could you explain more about your solution? Same problem here...

Thanks!