waveform80/picamera #682: Status of this project

iambutaman, 2021-03-15:
I have not seen any commits or merged MRS for quite some time now. What's the status of this project? Is it abandoned? Does the original author not have time anymore? Is it possible letting other developers push and merge code to this repo?

waveform80, 2021-03-21:
Sadly I haven't had time to do much with picamera over the past couple of years - Ubuntu on Pi (which became my day job two years ago), and piwheels (another open-source project I do a lot of work on) have swallowed pretty much all available time. That's beginning to ease a bit now (well, sort of -- I'm studiously ignoring the gaping hole that is arm64 support on piwheels because there's no time to deal with *that*), but I have recently found time to burn down the backlog on gpiozero (another project I do quite a bit on), and a few other minor projects. Unfortunately picamera requires a good block of attention, and I haven't got the time for that right now, but it's looking possible I *might* be able to later this year.

What might be in picamera's future? Given that picamera is built on MMAL, and that MMAL is now considered a "legacy" interface with all exciting future development going into libcamera, I'd say it's wise that people consider picamera "legacy" too. It's not going away anytime soon, but there's not going to be *much* new exciting stuff in any future 1.x release.

What might realistically be included? I'd urge people to go have a look at [picamerax](https://pypi.org/project/picamerax) as a more up to date alternative. Sadly I can't say I've had more than a skim of the fork but what I have skimmed looked incremental and sane, which is pretty much what one might reasonably expect of a future picamera 1.x.

On that point, I should probably seriously consider giving additional developers (like the picamerax devs) commit access but that in itself requires serious time for review of their work, and as noted I've only had a cursory skim.

TL;DR: not dead *yet*, but should definitely be considered "legacy", and it's certainly had an extended hiatus which *might* end later this year. Go check out [picamerax](https://pypi.org/project/picamerax) if you haven't already.

knabbers, 2021-08-17:
Please could picamerax be mentioned in the README "as a more up-to-date alternative"?
It's nice to know it's approximately endorsed by you, and might prevent people making unnecessary PRs to this repo?