waveform80/picamera #688: How to reduce capture time to 30ms?

jsib0, 2021-05-04:
The capture() method currently take ~1-1.15 seconds to output an image. With an image resolution goal of 4064x3040, how can I achieve a capture time of < 30ms?

pwr33, 2021-06-24:
try using 

use_video_port=True

in the capture statement, 

camera port is slower than the framerate, video port is faster, but at full framesize not sure how fast.

read the manual, it is very good, with lots of good examples.

jsib0, 2021-06-24:
@pwr33 

`use_video_port=True` is the obvious choice; it still does not achieve the desired time and quality we are looking for. Thank you for directing me to the manual. I should have been clear in my initial question. I was looking for something more of "out-of-the-box" that other may have tried.  I will close as this there seems to be no work-around.

closed 2021-06-24 by jsib0

pwr33, 2021-06-24:
think you may be hitting the basic transfer speed limit of the gpu, don't think that has been massively upgraded even on the pi4

maybe try the raspistill C code as a start point