waveform80/picamera #690: Display numpy array using picamera

watchever, 2021-06-04:
I really like the display get through start_preview, because it is simple and low level (so probably fast).
Is it also possible to display a fixe image like numpy or any other output through picamera api ? 
At present I use matplotlib but would really prefer to use picamera.

Thanks for your help

6by9, 2021-06-04:
3.14. Overlaying images on the preview
https://picamera.readthedocs.io/en/release-1.13/recipes1.html#overlaying-images-on-the-preview

watchever, 2021-06-04:
Thanks a lot for your reply, exactly what I need.

watchever, 2021-06-04:
Hu... Sorry another question I use add_overlay to display a new fixed image each n seconds. But after few iterations I got an Out of Memory error. In fact only the last overlay interest me. Is there a smart way to free memory ?
Following is the last exception line.

  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Unable to enable port vc.ril.camera:out:2: Out of memory