waveform80/picamera #697: picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': I/O error

Thebeefed, 2021-07-27:
I'm using Raspberry Pi 3 Model B+.

```vcgencmd get_camera``` gives me ```supported=1 detected=1```

```raspistill -o test.jpg``` gives me the test.jpg file but also gives this message
```
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
mmal: Failed to create camera_info component
```

Then, I try to run this python code below, but it gives me this error message
```
from picamera import PiCamera
camera = PiCamera()
```
Error:
```
mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 633, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': I/O error
```

Esser50K, 2021-08-13:
did you enable the camera yet? you can use `raspi-config` for that. The info is in the documentation though

nsayer, 2021-09-10:
I have this same problem, but instead of I/O error, I get Function not implemented. This was with a fully working script. apt-get upgrade broke it. I've opened another issue for this.

helioloureiro, 2021-11-09:
I upgrade to bullseye and got in the same situation.  Unfortunately it doesn't promising to get fixed soon.

https://raspberrypi.stackexchange.com/questions/132145/software-upgrade-broke-camera

helioloureiro, 2021-11-09:
My error is slightly different:

pi@raspberrypi3:~ $ ipython3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.20.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from picamera import PiCamera

In [2]: cam = PiCamera()
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
---------------------------------------------------------------------------
PiCameraMMALError                         Traceback (most recent call last)
<ipython-input-2-34374598ea04> in <module>
----> 1 cam = PiCamera()

/usr/lib/python3/dist-packages/picamera/camera.py in __init__(self, camera_num, stereo_mode, stereo_decimate, resolution, framerate, sensor_mode, led_pin, clock_mode, framerate_range)
    365         self._raw_format = 'yuv'
    366         self._image_effect_params = None
--> 367         with mo.MMALCameraInfo() as camera_info:
    368             info = camera_info.control.params[mmal.MMAL_PARAMETER_CAMERA_INFO]
    369             self._revision = 'ov5647'

/usr/lib/python3/dist-packages/picamera/mmalobj.py in __init__(self)
   2344 
   2345     def __init__(self):
-> 2346         super(MMALCameraInfo, self).__init__()
   2347         if PARAM_TYPES[mmal.MMAL_PARAMETER_CAMERA_INFO] is None:
   2348             found = False

/usr/lib/python3/dist-packages/picamera/mmalobj.py in __init__(self)
    629         super(MMALBaseComponent, self).__init__()
    630         self._component = ct.POINTER(mmal.MMAL_COMPONENT_T)()
--> 631         mmal_check(
    632             mmal.mmal_component_create(self.component_type, self._component),
    633             prefix="Failed to create MMAL component %s" % self.component_type)

/usr/lib/python3/dist-packages/picamera/exc.py in mmal_check(status, prefix)
    182     """
    183     if status != mmal.MMAL_SUCCESS:
--> 184         raise PiCameraMMALError(status, prefix)
    185 

PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented

6by9, 2021-11-09:
With regard Bullseye (released on 7th November, so TOTALLY unrelated to this issue), the blog post about the release is fairly plain.
https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/
> New camera driver
>
> The driver used by Raspberry Pi to access camera modules has now been replaced with libcamera, a standard Linux API. As with the video driver change, this means less closed-source proprietary code, and makes it easier for third parties to develop new camera hardware and software.
>
> For anyone who has written camera-based applications in the past, this is a very significant change, and is too large to be covered here. There will be a blog post dedicated to the new camera driver published soon, so watch this space!

helioloureiro, 2021-11-09:
@6by9 I was going to mention that I got it working after install libcamera-tools and libcamera-apps.

So the picamera package has wrong dependencies.

6by9, 2021-11-09:
libcamera-tools and libcamera-apps have no relation to picamera - they will not reinstall the MMAL dependencies which picamera relies on.

The legacy camera stack (including picamera) is not officially supported on Bullseye.

helioloureiro, 2021-11-09:
So @6by9 what would be fix for that?  I may missed that point.  It works via libcamera-tools, but remains complaining w/ "'vc.camera_info': Function not implemented".

I can see it is defined at header files:

$ grep -r camera_info /usr/include/interface/mmal/
/usr/include/interface/mmal/util/mmal_default_components.h:# define MMAL_COMPONENT_DEFAULT_CAMERA_INFO      "camera_info"
/usr/include/interface/mmal/util/mmal_default_components.h:# define MMAL_COMPONENT_DEFAULT_CAMERA_INFO      "vc.camera_info"

nsayer, 2021-11-09:
So this means that picamera either needs to be rewritten against libcamera or it becomes abandonware.

If I understand the announcement correctly, it does imply that if the work were to get done it would mean that picamera would be applicable for platforms beyond the Raspberry Pi because libcamera is apparently a standard facility across multiple Linux variants.

6by9, 2021-11-09:
You can still enable the legacy stack, but the default is now libcamera. https://forums.raspberrypi.com/viewtopic.php?t=323390

Python bindings for libcamera are already in the works.

danielfaust, 2021-11-12:
@6by9 Thank you for posting the link to the topic in the Raspberry Pi forums. It enabled me to reuse the old stack.

I think that this is a big issue which doesn't get enough visibility. Googling for answers does barely yield any useful results, and the news post [0] from the foundation about this changes should give the user hints to how to re-enable the old behavior.

There are things the new stack can't do (at this time) which might break old code, in my case it was `picamera`, on which I relied to stream the video with a current timestamp and preform image snapshots in parallel at a given time interval.

In any case, I will copy your code into this issue, should the forum become inaccessible:

> Edit /boot/config.txt, remove the line "camera_auto_detect=1", and add "start_x=1" and "gpu_mem=128". Rebooting at this stage will reload the old V4L2 driver. Stop at this point if that is sufficient for your needs.

> If you need raspistill/vid, then build the userland libraries for yourself.

```
cd ~
sudo apt install cmake
mkdir bin
git clone https://github.com/raspberrypi/userland
cd userland
./buildme
cp build/bin/* ~/bin/
```

> If a "bin" directory exists in your home directory, then the default .profile adds it to your path, hence it will find these copies of the binaries.
If you need the binary accessible by all users, then copying them into /usr/local/bin would be the normal solution.

> Note that with the graphics stack having been changed to be Full KMS by default, then the firmware can no longer a a preview display to the screen. Please use the -n option (for no preview), or edit /boot/config.txt again and change "dtoverlay=vc4-kms-v3d" to "dtoverlay=vc4-fkms-v3d".
Second note, this is still 32bit only. MMAL support on 64bit is still partially broken.

[0] https://www.raspberrypi.com/news/an-open-source-camera-stack-for-raspberry-pi-using-libcamera/

nsayer, 2021-11-12:
That's not enough. If you attempt this with the current libraspberrypi0/bootloader packages, what you get is "Failed to create MMAL component b'vc.camera_info': Function Not Implemented." You have to backtrack those packages to the 20210831-1 versions for now.

danielfaust, 2021-11-12:
> You have to backtrack those packages to the 20210831-1 versions for now.
What does that mean? How do I do that?

nsayer, 2021-11-12:
Go to http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D

Download 

libraspberrypi0_1.20210831-1_armhf.deb
libraspberrypi-bin_1.20210831-1_armhf.deb
libraspberrypi-dev_1.20210831-1_armhf.deb
libraspberrypi-doc_1.20210831-1_armhf.deb
raspberrypi-bootloader_1.20210831-1_armhf.deb

as root, do "dpkg -i " for all those files. You may need to remove vcdbg or make other package adjustments so that those fit.

Afterwards, do "apt-mark hold" for all of them so that future apt upgrades don't overwrite them until that bug gets fixed.

danielfaust, 2021-11-21:
@nsayer How safe is this downgrading? Does it have side-effects? Also, can it be completely reverted by "unholding" them and then issuing an `apt-get update & apt-get upgrade`?

nsayer, 2021-11-21:
Well, it worked fine for me.

And, yes, un-holding and upgrading is sufficient to undo this.

danielfaust, 2021-11-22:
This is what worked for me:

```
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20210831-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20210831-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20210831-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-dev_1.20210831-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-doc_1.20210831-1_armhf.deb

sudo apt purge vcdbg
sudo dpkg -i raspberrypi-bootloader_1.20210831-1_armhf.deb
sudo dpkg -i libraspberrypi0_1.20210831-1_armhf.deb
sudo dpkg -i libraspberrypi-bin_1.20210831-1_armhf.deb
sudo dpkg -i libraspberrypi-dev_1.20210831-1_armhf.deb
sudo dpkg -i libraspberrypi-doc_1.20210831-1_armhf.deb

sudo apt-mark hold raspberrypi-bootloader_1.20210831-1_armhf.deb
sudo apt-mark hold libraspberrypi0_1.20210831-1_armhf.deb
sudo apt-mark hold libraspberrypi-bin_1.20210831-1_armhf.deb
sudo apt-mark hold libraspberrypi-dev_1.20210831-1_armhf.deb
sudo apt-mark hold libraspberrypi-doc_1.20210831-1_armhf.deb
```

`vcdbg` needs to be removed because it is included in `libraspberrypi-bin`, else `libraspberrypi-bin` would not install.

But I still have a question: doing an `apt search raspberrypi` yields:
```
raspberrypi-bootloader/stable 1:1.20211029-1 armhf [upgradable from: 1:1.20210831-1] -- Raspberry Pi bootloader
libraspberrypi0/stable 1:2+git20210928~141951+6e8f786-2 armhf [upgradable from: 1:1.20210831-1] -- EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV
libraspberrypi-bin/stable 1:2+git20210928~141951+6e8f786-2 armhf [upgradable from: 1:1.20210831-1] -- Miscellaneous Raspberry Pi utilities
libraspberrypi-dev/stable 1:2+git20210928~141951+6e8f786-2 armhf [upgradable from: 1:1.20210831-1] -- EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
libraspberrypi-doc/stable 1:2+git20210928~141951+6e8f786-2 armhf [upgradable from: 1:1.20210831-1] -- EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)

firmware-brcm80211/stable,now 1:20210315-3+rpt3 all [installed] -- Binary firmware for Broadcom/Cypress 802.11 wireless cards
libraspberrypi-bin-dbgsym/stable 1:2+git20210928~141951+6e8f786-2 armhf -- debug symbols for libraspberrypi-bin
libraspberrypi0-dbgsym/stable 1:2+git20210928~141951+6e8f786-2 armhf -- debug symbols for libraspberrypi0
raspberrypi-kernel/stable,now 1:1.20211029-1 armhf [installed] -- Raspberry Pi bootloader
raspberrypi-kernel-headers/stable 1:1.20211029-1 armhf -- Header files for the Raspberry Pi Linux kernel
raspberrypi-net-mods/stable,now 1.3.2 all [installed] -- Network configuration for the Raspberry Pi UI
raspberrypi-sys-mods/stable,now 20211005+bullseye armhf [installed] -- System tweaks for the Raspberry Pi
raspberrypi-ui-mods/stable 1.20211110 armhf -- Config to customise the LXDE UI for the Raspberry Pi
```

So I'm wondering how safe it is to leave the latter not-upgradeable entries at the newest version, specially `libraspberrypi-bin-dbgsym` and `libraspberrypi0-dbgsym`

In any case, thank you for your help, because now it works.

Another question: `libraspberrypi-bin` contains `/usr/bin/raspistill` and `/usr/bin/raspivid`, at least according to https://packages.ubuntu.com/focal-updates/arm64/libraspberrypi-bin/filelist

So, could it be that it is not necessary to compile and install userland as described above ( where `git clone https://github.com/raspberrypi/userland` is mentioned )?

BlackLight, 2021-11-28:
@6by9 

> For anyone who has written camera-based applications in the past, this is a very significant change, and is too large to be covered here. There will be a blog post dedicated to the new camera driver published soon, so watch this space!

I really don't believe that this is the right approach from the RPi Foundation to communicate such a big change, and I understand why many are frustrated with it.

There are _thousands_ of scripts and examples out there that reference `picamera`, and they have been written over many years. I have built many pieces of automation that leverage `picamera` to stream and record video between devices, and they've all been broken by the bullseye upgrade.

You can't get away with a random blog post that says "hey guys, FYI we've decided to completely kill the old picamera API, we'll let you know as soon as we have time how you are supposed to migrate your scripts". This is the best way of getting developers pissed, and it also has an unfortunate precedent (the way the omxplayer layer has been left to break and rot on its own without viable alternatives). What I would have expected:

- The `picamera` relevant pages (Github, documentation and pypi) should have reported, in big capital letters and before bullseye was released as a new stable, _WARNING: This library is deprecated and is going to break in bullseye, [link to migration steps here]_.
- A warning upon pip/apt install/upgrade (or import) of `picamera` on bullseye that clearly warns the user about the deprecation.
- A compatibility layer: the v4l2 API is widely used and documented, but the `picamera` module offers functionalities that make it easier to e.g. start a video stream or record with a preview even for a beginner, without having to mangle with ffmpeg/gstreamer/v4l2/gtk. The `picamera` module could have simply relied on e.g. a gstreamer wrapper instead of the proprietary interface, while providing the same interface as before to maintain back-compatibility with previous software.
- A viable alternative: I've given the Pi camera a try with my ffmpeg and gstreamer Python streamers, and the performance is not at all comparable. This reminds me, again, about the messy deprecation of omxplayer: "omxplayer is falling apart, don't use it anymore, use vlc instead, but we are also aware that vlc is not there yet when it comes to RPi integration". When you deprecate something, make sure that you communicate it properly, and that you provide a satisfying alternative to the thing you are deprecating.

nsayer, 2021-11-28:
Raspberry Pi is becoming low-rent Apple.

t1u1, 2022-01-07:
> Python bindings for libcamera are already in the works.

Can we follow progress for this somewhere? Is there a code repo? Thanks!

6by9, 2022-01-07:
https://forums.raspberrypi.com/viewtopic.php?p=1955918

lexodistro, 2022-01-25:
bro 

In order to get the piCamera modules not to throw errors I had to manually update my repositories for bullseye this is important to do or else you will get errors like so 

mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)

To fix this run these commands to update the repositories and remove the vlc errors

```
sudo apt-get remove --purge vlc-nox
sudo apt-get autoclean
sudo dpkg --configure -a
sudo apt-get -f install
sudo apt-get autoremove
sudo apt-get update
sudo apt-get dist-upgrade
```

Then make sure to reenable the legacy camera interface

`sudo rasps-config `

Then enable it 

After this do a reboot 

`sudo reboot`

Then send me lots of thanks its quick and dirty but you can run PiCamera on bullseye and then update to Libcamera when they release picamer2 with the python3 bindings!!! be well and hope this helped someone.

Kevin1299, 2022-01-25:
@lexodistro I think it fixed the problem for me thank you :) However I am not able to use the VNC desktop anymore. Do you have an idea how I can fix this?

Wolfhound905, 2022-01-25:
@lexodistro thanks

gkzsolt, 2022-01-27:
A million thanks, @lexodistro ! This saved my day :-)  I completely agree with BlackLight, btw.

worldofchris, 2022-01-28:
@lexodistro you have saved the day. I was working against the clock to get something ready for an after school club when I hit this issue.

One minor issue. I think you have a typo in your post:
```
sudo rasps-config
sudo: rasps-config: command not found
```
Should it be:
```
sudo raspi-config
```

?

Thank you SO much though.

danielfaust, 2022-01-31:
Just a note on @lexodistro 's solution.

According to https://downloads.raspberrypi.org/raspios_armhf/release_notes.txt the 2022-01-28 release re-added support for the old camera stack ("Legacy camera applications and libraries reinstalled") and `raspi-config` got a new option to enable the legacy stack `raspi-config - add option to switch to legacy camera mode`.

This means that it is enough to issue a `sudo apt update; sudo apt dist-upgrade` and after the reboot issue `sudo raspi-config` to enable the legacy camera stack in the `` section.

If you were holding the old packages you'd need to unhold them before doing the `dist-upgrade`:

```sudo apt-mark unhold raspberrypi-bootloader
sudo apt-mark unhold libraspberrypi0
sudo apt-mark unhold libraspberrypi-bin
sudo apt-mark unhold libraspberrypi-dev
sudo apt-mark unhold libraspberrypi-doc

sudo apt update; sudo apt upgrade```

and then after the reboot issue the `sudo apt dist-upgrade`