waveform80/picamera #698: Documentation: Minimum exposure time does not seem to make sense

edbird, 2021-07-27:
I am confused by 6.1.3.1. Minimum exposure time from the documentation. The explanation doesn't seem to make any sense.

Here is what it says:

> There are naturally limits to the minimum exposure time: reading out a line of elements must take a certain minimum time. For example, if there are 500 rows on our hypothetical sensor, and reading each row takes a minimum of 20ns then it will take a minimum of 500 \times 20\text{ns} = 10\text{ms} to read a full frame. This is the minimum exposure time of our hypothetical sensor.

Given what is written in the previous section, which describes the reading of the sensor as a process of "reading an entire row of sensor cells" and "resetting an entire row of sensor cells", the above doesn't make any sense. Presumably the minimum exposure time is the time taken between row operations. In other words, if there is a clock signal which controls the reading of a row, and the clock is 20ns then the minimum exposure time is 20ns, the same as the "row reading clock".

Put another way, if the idea of introducing a clock into the explanation was unhelpful, there is clearly some minimum time to "reset a row" or "read a row". I assume that "reading a row" takes longer than "resetting a row". (But maybe I am wrong.)

The minimum exposure time is whichever takes longer "reading a row" or "resetting a row".

This then brings us to a second issue. Why I can see why "resetting a row" (discharging a bank of capacitors) would be a fairly easy operation to implement in hardware as a simultanious operation for the whole row, I can't exactly see how the reading of a whole row can be done in parallel unless the entire row contains an ADC for each light sensing element, which although possible seems unlikely and an expensive design.

Is it really correct that a row-reading operation is done in parallel?

edbird, 2021-07-27:
To further clarify my first point:

The minimum exposure time must be this:

- Reset row 0
- Next operation: read the row which was just reset (row 0)

Whereas a longer exposure time would be obtained from this sequence:

- Reset row 0
- Wait some period of time (reset row 1, reset row 2, reset row 3, ..., reset row N)
- Read row 0 (simultaniously: reset row N + 1)

6by9, 2021-07-27:
The docs are incorrect in their description of operation.
Most sensors will have a minimum value that can be programmed into the exposure time register, but it is typically < 10 lines, and is not the frame length.
The frame length * row readout time does dictate the max frame rate.

Whether the entire row is read in parallel or not depends on the sensor. What is more likely to happen is that the charge from a row is dropped into a store block at the start of the line readout, and then the ADC reads each pixel in turn from there. This avoids pixels further along the line continuing to be exposed whilst the line is being read out. Having only 1 ADC avoids variation between multiple units.

Switch to a global shutter sensor and the entire sensor array is dropped into a parallel array at the start of frame, so all pixels are exposed for exactly the same period of time and readout can then be completed at leisure.

edbird, 2021-07-27:
Ok that makes sense. So it is even simpler than the documentation makes out. Simply put there is a minimum exposure time dictated by valid values which can be stored in a register, and that register controls the number of cycles between reseting a row and reading a row.

Thanks for the clarification.