waveform80/picamera #701: I-Frame Extraction

rossGardiner, 2021-08-31:
Is it possible to extract keyframes (iframes) at the encoding stage of the video stream? I have an application which requires iframes from specific points in a H264 stream, selecting iframes from the stream in post (using ffmpeg) takes too long. Could picamera output iframes as they are encoded?

Any advice appreciated!

6by9, 2021-08-31:
The GPU gives you flags alongside each encoded buffer.
With the Encoder class you get [```_callback```](https://picamera.readthedocs.io/en/release-1.13/api_encoders.html#picamera.PiEncoder._callback) which gives you a [```MMALBuffer```](https://picamera.readthedocs.io/en/release-1.13/api_mmalobj.html#picamera.mmalobj.MMALBuffer) object. 
The [```flags``` member](https://picamera.readthedocs.io/en/release-1.13/api_mmalobj.html#picamera.mmalobj.MMALBuffer.flags) will have ```MMAL_BUFFER_HEADER_FLAG_KEYFRAME``` set for an I-frame.

rossGardiner, 2021-09-02:
Thank you, this was exactly what I needed.

closed 2021-09-02 by rossGardiner