waveform80/picamera #705: picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented

nsayer, 2021-09-10:
I have picamera 1.13 installed. I had a perfectly working webcam script running until I did an apt-get upgrade today. That upgraded did this:

```
Get:1 http://archive.raspberrypi.org/debian bullseye/main armhf libc6-dbg armhf 2.31-13+rpt2+rpi1 [6377 kB]
Get:2 http://archive.raspberrypi.org/debian bullseye/main armhf libc-devtools armhf 2.31-13+rpt2+rpi1 [245 kB]
Get:3 http://archive.raspberrypi.org/debian bullseye/main armhf libc6-dev armhf 2.31-13+rpt2+rpi1 [1813 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye/main armhf libc-dev-bin armhf 2.31-13+rpt2+rpi1 [266 kB]
Get:5 http://archive.raspberrypi.org/debian bullseye/main armhf libc6 armhf 2.31-13+rpt2+rpi1 [2297 kB]
Get:6 http://archive.raspberrypi.org/debian bullseye/main armhf libc-bin armhf 2.31-13+rpt2+rpi1 [682 kB]
Get:7 http://archive.raspberrypi.org/debian bullseye/main armhf bluez armhf 5.55-3.1+rpt1 [843 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf libc-l10n all 2.31-13+rpt2+rpi1 [863 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main armhf locales all 2.31-13+rpt2+rpi1 [4084 kB]
Get:10 http://archive.raspberrypi.org/debian bullseye/main armhf libraspberrypi-dev armhf 1:2+git20210719~113532+97bc818-3 [180 kB]
Get:11 http://archive.raspberrypi.org/debian bullseye/main armhf libraspberrypi-doc all 1:2+git20210719~113532+97bc818-3 [1564 B]
Get:12 http://archive.raspberrypi.org/debian bullseye/main armhf libraspberrypi-bin armhf 1:2+git20210719~113532+97bc818-3 [40.8 kB]
Get:13 http://archive.raspberrypi.org/debian bullseye/main armhf libraspberrypi0 armhf 1:2+git20210719~113532+97bc818-3 [177 kB]
```

After rebooting, now the script does this:

```
mmal: mmal_component_create_core: could not find component 'vc.camera_info'
Traceback (most recent call last):
  File "/home/nsayer/./cam.py", line 85, in <module>
    with picamera.PiCamera(resolution='1600x1200', framerate=24) as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 367, in __init__
    with mo.MMALCameraInfo() as camera_info:
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2346, in __init__
    super(MMALCameraInfo, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': Function not implemented
```

TO REITERATE: _NOTHING_ changed other than the package upgrades noted above.

danielfaust, 2021-11-11:
I have the same issue on a fresh install of bullseye.

nsayer, 2021-11-11:
I brought this up on the Raspberry Pi forums and after a lot of pushback and arguing I believe I *finally* got an acknowledgment from a Pi engineer that this is a bug that he says he will fix.

6by9, 2021-11-11:
Duplicate of #697

(Guess what 6by9 here and commenting on 697 is the same Pi engineer 6by9 on the forums. Being grumpy/sarcastic about *finally* getting a response doesn't promote your case).

closed 2021-11-11 by 6by9

nsayer, 2021-11-11:
I posted about this problem TWO MONTHS AGO on the pi forums as well and got nothing but pushback and attitude about it. I have every justification for being grumpy about this.

danielfaust, 2021-11-12:
@6by9 Thank you for linking the duplicate. The link to the forum topic posted in there helped me to re-enable the old behavior.

istvanzk, 2021-11-21:
I have the same problem on a RPi A+ running  bullseye and picamera (from within venv). Any updates or work around we can use until it gets fixed? Thank you.

nsayer, 2021-11-21:
The workaround is to fetch these packages and downgrade to them:

libraspberrypi0_1.20210831-1_armhf.deb
libraspberrypi-bin_1.20210831-1_armhf.deb
libraspberrypi-dev_1.20210831-1_armhf.deb
libraspberrypi-doc_1.20210831-1_armhf.deb
raspberrypi-bootloader_1.20210831-1_armhf.deb

Adenialzz, 2022-01-02:
> 

Hi bro, thanks for finding a workaround eventually, but these packages don’t seem to work on arm64 architecture. So for Raspberry Pi 4b, do you know what to do?

6by9, 2022-01-02:
MMAL is not supported under 64bit userland - https://github.com/raspberrypi/userland/issues/688

Adenialzz, 2022-01-03:
> MMAL is not supported under 64bit userland - [raspberrypi/userland#688](https://github.com/raspberrypi/userland/issues/688)

thank you for your reply, does that mean picamera is not available on Raspberry Pi 4b?

6by9, 2022-01-04:
Picamera works fine on Pi4 32bit userland, which the Pi4 runs quite happily.
With Raspberry Pi OS Bullseye release you will need to enable the legacy camera stack.