waveform80/picamera #716: libmmal.so: cannot open shared object file: No such file or directory

harshavardhannarla, 2022-02-27:
Iam unable to import picamera module , It is giving error :libmmal.so: cannot open shared object file: No such file or directory

XPbIM3, 2022-03-10:
Same with me.
-Raspberry Pi 4B 4Gb
-64Bit rasp os
-libmmal.so is present in /opt/vc/lib/libmmal.so but seem to be a 32bit binary

6by9, 2022-03-10:
MMAL (which picamera uses) is not supported on 64bit userland
https://github.com/raspberrypi/userland/issues/688

closed 2022-03-10 by 6by9

DUCHUAN99, 2023-11-06:
I use a 32bit operating system on raspberry 3b+ and still encounter the above error, what should I do?

Swarkin, 2023-11-11:
I have the same issue on 32bit

VisheshSingh1, 2024-01-15:
did you guys got any solution .. I am getting same error .. using model 3 b os 32bit

Liuss44, 2024-01-17:
I had same error and I solved it by moving

/opt/vc/lib/libmmal_vc_client.so
/opt/vc/lib/libmmal_util.so
/opt/vc/lib/libmmal_core.so
/opt/vc/lib/libmmal_components.so
/opt/vc/lib/libmmal.so
into /usr/lib/

If it can`t find any other files just use
`sudo find / -name "name of the file"`
and move then into /usr/lib/ worked for me

hope this helps. :)