waveform80/picamera #719: shutter_speed TypeError

Bra1nsen, 2022-05-26:
Dear Reader,

I would like to **set the shutter speed below 1m**s - for example 0.1ms (which i know is possible with the hq cam)

therefore a **change in the source code** is nessecary (solve TypeError). Are there any instructions on how to do that?

I would be infinitly grateful for help.

```
Traceback (most recent call last):
  File "/home/pi/kotzmichan.py", line 39, in <module>
    camera.shutter_speed = 0.5
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 2594, in _set_shutter_speed
    self._camera.control.params[mmal.MMAL_PARAMETER_SHUTTER_SPEED] = value
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 1493, in __setitem__
    func(self._port, key, conv(value)),
ctypes.ArgumentError: argument 3: <class 'TypeError'>: wrong type

```

6by9, 2022-05-26:
> Any chance to set shutter speed below 1ms?

Not with the firmware stack (ie raspistill/vid or picamera).

Bra1nsen, 2022-05-26:
What would you recommend me?

6by9, 2022-05-26:
> What would you recommend me?

Already stated in https://github.com/waveform80/picamera/issues/719#issuecomment-1138513164
> It should be possible with libcamera.

```libcamera-hello --shutter 100``` should request a 100usec exposure time.

closed 2022-05-26 by 6by9

Bra1nsen, 2022-05-26:
Iam sure youre just incapable :)

6by9, 2022-05-26:
The legacy camera stack is deprecated.
There will not be further significant changes made to it, and therefore supporting sub-millisecond values for exposure time will not happen.

Bra1nsen, 2022-05-26:
`libcamera-jpeg -o ff01.jpg --shutter 0.001`
[image]

this is supposed to be the lowest shutter speed.. that I don't laugh, that's bullshit sorry

Bra1nsen, 2022-05-26:
just to compare whats possible:
[image]

6by9, 2022-05-26:
As you can't keep your language clean, this will be my last comment on this issue.

https://www.raspberrypi.com/documentation/accessories/camera.html#exposure-control
It's an integer. Min exposure is therefore 1usec. 0 means auto exposure. The text output of libcamera-jpeg will have told you the value applied.
```
pi@raspberrypi:~ $ libcamera-jpeg --shutter 1 -o foo.jpg
Made DRM preview window
[0:00:53.362209837] [784]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3548-a11d63f9
[0:00:53.413879393] [785]  WARN RPI raspberrypi.cpp:1237 Mismatch between Unicam and CamHelper for embedded data usage!
[0:00:53.414545893] [785]  INFO RPI raspberrypi.cpp:1352 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:00:53.415356745] [784]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:00:53.415842430] [785]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:00:53.437070911] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Saturation = 1.000000
[0:00:53.437196374] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Contrast = 1.000000
[0:00:53.437265615] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Brightness = 0.000000
[0:00:53.437327226] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AwbMode = 0
[0:00:53.437384134] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Sharpness = 1.000000
[0:00:53.437448689] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureValue = 0.000000
[0:00:53.437510393] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeExposureMode = 0
[0:00:53.437564837] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureTime = 1 <<<<<<<<<<<<<<<<<<<
[0:00:53.437619819] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:00:53.437678893] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeMeteringMode = 0
[0:00:53.437737689] [789]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: NoiseReductionMode = 3
[0:00:58.556495951] [784]  INFO Camera camera.cpp:1029 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[0:00:58.558325363] [785]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[0:00:58.639077922] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Saturation = 1.000000
[0:00:58.639220992] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Contrast = 1.000000
[0:00:58.639279305] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Brightness = 0.000000
[0:00:58.639313378] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AwbMode = 0
[0:00:58.639344507] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Sharpness = 1.000000
[0:00:58.639391765] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureValue = 0.000000
[0:00:58.639426078] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeExposureMode = 0
[0:00:58.639453800] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureTime = 1 <<<<<<<<<<<<<<<<<<<
[0:00:58.639483095] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[0:00:58.639517780] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeMeteringMode = 0
[0:00:58.639548779] [792]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: NoiseReductionMode = 2
Still capture image received
```
vs
```
pi@raspberrypi:~ $ libcamera-jpeg --shutter 0.01 -o foo1.jpg
Made DRM preview window
[0:01:05.057698975] [794]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3548-a11d63f9
[0:01:05.097133351] [795]  WARN RPI raspberrypi.cpp:1237 Mismatch between Unicam and CamHelper for embedded data usage!
[0:01:05.097802023] [795]  INFO RPI raspberrypi.cpp:1352 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:01:05.098559287] [794]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:01:05.099191052] [795]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:01:05.120977543] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Saturation = 1.000000
[0:01:05.121317592] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Contrast = 1.000000
[0:01:05.121526607] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Brightness = 0.000000
[0:01:05.121726547] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AwbMode = 0
[0:01:05.121928247] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Sharpness = 1.000000
[0:01:05.122151484] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureValue = 0.000000
[0:01:05.122353609] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeExposureMode = 0
[0:01:05.122552383] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureTime = 0 <<<<<<<<<<<<<<<<<<
[0:01:05.122748713] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:01:05.122947079] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeMeteringMode = 0
[0:01:05.123145113] [799]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: NoiseReductionMode = 3
[0:01:10.240590070] [794]  INFO Camera camera.cpp:1029 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[... cut ...]
[0:01:10.318977354] [802]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeExposureMode = 0
[0:01:10.319037483] [802]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureTime = 0 <<<<<<<<<<<<<<<<<<<
[0:01:10.319096093] [802]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[0:01:10.319166481] [802]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeMeteringMode = 0
[0:01:10.319227036] [802]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: NoiseReductionMode = 2
Still capture image received
```

Bra1nsen, 2022-05-26:
Thanks for the info about autoexpo, I didn't know that.

 **I ask how the exposure time can be lowered further.** You recommend libcamera to me ("libcamera-hello --shutter 10") and close mine please. That does **not help** at all if I can only go down to a minimum of 1 here. **Just don't answer if you don't have a solution**

6by9, 2022-05-26:
Locked as this is no longer relevant to Picamera, and you appear to be ignoring the fact that Picamera/MMAL take MILLISECONDS and libcamera takes MICROSECONDS, hence reducing the minimum by a factor of 1000.

Please adjust your attitude if you wish to ask for assistance in future.