waveform80/picamera #735: Error : OSError: libbcm_host.so: cannot open shared object file: No such file or directory

lazyboytql, 2024-02-29:
Currently, when I run the picamera program in the virtual environment (env), this error is displayed:
Traceback (most recent call last):
  File "/home/linh/docker-python/camera.py", line 1, in <module>
    import picamera
  File "/home/linh/myenv/lib/python3.11/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/home/linh/myenv/lib/python3.11/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/home/linh/myenv/lib/python3.11/site-packages/picamera/mmal.py", line 47, in <module>
    from .bcm_host import VCOS_UNSIGNED
  File "/home/linh/myenv/lib/python3.11/site-packages/picamera/bcm_host.py", line 47, in <module>
    _lib = ct.CDLL('libbcm_host.so')
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libbcm_host.so: cannot open shared object file: No such file or directory

In this virtual environment I have installed all the necessary libraries to run the program.
I am using raspberry pi 4 Model B+ and Picamera V3
Please help me fix this error and get your response as soon as possible, thank you very much !!
[image]

guoxiaolong5669, 2024-03-21:
I have the same problem and do not know how to solve it

Vaylordqt, 2024-03-22:
did you resolve this issue ?

imbrianj, 2024-04-11:
Are you running a 32 bit or 64 bit OS?

6by9, 2024-04-15:
Looking at the OP

> I am using raspberry pi 4 Model B+ and Picamera V3

The v3 camera is ONLY supported by libcamera, not the legacy firmware camera stack. picamera is therefore incompatible with it as it is using the legacy stack. You need to use libcamera / rpicam-apps / picamera2.

If using a V1, V2, or HQ camera, then picamera can only work on the 32 bit OS, even if the SoC is capable of running a 64bit one.

The legacy stack is deprecated, and you are strongly advised to migrate to a libcamera based solution.